Ghostkeeper/X3GWriter #3: Plugin assumes that printer is a Replicator 1 Dual

d235j, 2017-10-29:
I've been working on a proper profile for the Replicator 2. Unfortunately though, X3GWriter assumes that the printer is a Replicator 1 Dual, which has a different steps/mm parameter and therefore produces incorrect X3G files.

Ideally it would be possible to provide parameters for preferred/require plugins directly in the printer definition file; another thing that I put together is a PostProcessing script that inserts percentage M codes into the output gcode file.

d235j, 2017-10-29:
Additionally, the plugin passes the printer type to GPX twice, once via config file, and again on the command line, at https://github.com/Ghostkeeper/X3GWriter/blob/master/X3GWriter.py#L64.

d235j, 2017-10-29:
To assist with this: is there a way for a printer definition to provide custom parameters that a plugin could read? This would be very helpful here.

d235j, 2017-10-30:
Hmm — looking at https://github.com/markwal/GPX/blob/master/src/gpx/gpx.c#L4092 it seems like it would be possible to generate a gpx .ini file on the fly using printer definition parameters, and then feed it to gpx. This would be significantly more extensible than having a single parameter in the printer definition to select the hardcoded printer type in GPX — but would also be more work in the plugin.

Ghostkeeper, 2017-10-30:
I talked with d235j via IRC a bit. Sorry that I'm still a bit behind on my Github notifications (read: 1175 notifications atm).

When you run GPX from a command line it'll tell you that you can run it with `-c <ini file>` to load a custom configuration. It also gives some examples of such INI files, such as this one: https://github.com/markwal/GPX/blob/master/examples/example-machine.ini

It's not practical to put such an INI file in the machine definition itself, even though it belongs there.

I also don't think it's practical to actually define settings for each of GPX's parameters. It can be done, certainly, from this plug-in even. You'd look up the fdmprinter definition container and add some setting definitions there. A printer definition could then override some of those settings and the settings can be retrieved again when starting GPX. But that would mean that these printers simply fail to load at all (with an error message that the setting doesn't exist) when the X3GWriter plug-in is not present. And I don't know if we'd want that to happen. Perhaps once the container provider plug-in type is made we can facilitate this by letting this plug-in provide the .def.json files.

I have two options that I do think are practical:
1. Add a metadata field to the .def.json files that indicates the `machine_type` property (or what is currently in the `-m` parameter). This means that if someone defines a printer, they'd have to provide a definition file to Cura and add a preset to GPX. I don't think that it's that bad to have it "hardcoded" in GPX. It's going to have to be added either there or here.
2. Add a metadata field to the .def.json files that points to a .ini file inside the X3GWriter plug-in. This means that if someone defines a printer, they'd have to provide a definition file to Cura and add a preset to X3GWriter. I think adding the preset to GPX would be better since it'll be available to a wider audience then. However this also allows X3GWriter to override some of those settings with the data that it already has, such as the nozzle size and number of extruders (should the user want/be able to modify those).

Which do you think is better?

d235j, 2017-10-30:
> But that would mean that these printers simply fail to load at all (with an error message that the setting doesn't exist) when the X3GWriter plug-in is not present. 

Preferably when the X3GWriter plug-in is not present, the slicer would work as usual and generate gcode (which then could be converted with GPX or another tool, e.g. fed to Octoprint and converted by GPX on the print server).
Does the printer definition system not allow custom parameters that are ignored if not queried/defined by the plug-in? I've attempted to add new properties with `default_value=` and they seem to just have been ignored, with the printer definitions still loading.

That said, looking at that example config file, it seems that all the parameters needed are already in the printer .def.json except for the `steps_per_mm=` and `endstop=` values.
Some of the parameters are confusing, such as why is `has_heated_build_platform=` per tool.

What I'd prefer is some method that doesn't require having printer metadata in two different files (the .def.json and the .ini). Being able to generate .ini files to contribute back to GPX would definitely be useful, but the idea is that all of the state information about a particular printer be in one place or set of files, and not spread around.

> You'd look up the fdmprinter definition container and add some setting definitions there.

How would one go about doing this code-wise?

Ghostkeeper, 2017-10-30:
> > You'd look up the fdmprinter definition container and add some setting definitions there.
> 
> How would one go about doing this code-wise?

I was imagining something like:

    ContainerRegistry.getInstance().findDefinitionContainers(id = "fdmprinter")[0].addDefinition(...)

However it seems that there is no such `addDefinition` function. It may be possible to add some new definitions in the .def.json files themselves by adding them to the "values" category instead of "overrides". Perhaps that would be an even better solution overall.

Ghostkeeper, 2017-12-12:
This issue is currently waiting for a judgement on https://github.com/Ultimaker/Cura/pull/2960
The solution is ready on the machine_settings branch.

rpavlik, 2018-02-16:
Didn't realize this issue was here before I made a solution for myself (thanks for the start at the r2x def files, @d235j !), but I've been using the plugin patched with #7 for some time and it works well.

(FWIW, my corresponding definition files, forked from @d235j , I've finally uploaded here: https://gist.github.com/rpavlik/0fbc5289b2387dfe374a3ab003186845 )

mickletown, 2018-05-19:
Just tried following instructions for Cura 2.7.0, but until I used the extensions tag in Cura and selected X3G plug-in for printing, it would not register. Now awaiting output from printer to see if it worked!

Ghostkeeper, 2018-05-25:
This has now been fixed by https://github.com/Ghostkeeper/X3GWriter/pull/9 and https://github.com/Ghostkeeper/X3GWriter/pull/8.

Ghostkeeper, 2018-05-25:
The issue reported by Mickletown seems to be a different bug.

closed 2018-05-25 by Ghostkeeper