Ghostkeeper/X3GWriter #5: Prints centering on rear-right corner of build plate

stuartpb, 2018-01-16:
I'm using the X3GWriter plugin in Cura to output x3g, and whenever I try to run one of these output files, the print exhibits the kind of mis-centering behavior described by @markwal in [this FlashForge Google Groups thread](https://groups.google.com/d/msg/flashforge/iBHocaDUes4/ngmg_1LqAAAJ).

(My printer is a Replicator 1 Dual compatible Qidi Tech I running on Sailfish v7.8 r01234. I'm running Cura 3.1.0 with version 2.6.0 of this plugin.)

I've got everything configured correctly in my settings - the build plate is marked as having its origin at the center of the build plate, and the startup G-Code calls `M132 X Y Z` after zeroing the axes (I'm using [these settings](https://github.com/stuartpb/repligreater/tree/master/settings/qidi-tech-i/cura) as a base.)

For completion's sake, my current start code in use, exactly as defined:

```gcode
;@printer r1d

M73 P0 ; tell the firmware we're starting a build
G130 X118 Y118 A118 B118 ; set vref's to default_value

T0 ; home on the right extruder
G90 ;  set positioning to absolute
G28 X Y Z ; home XYZ at default_value home feedrates
G92 X0 Y0 Z0 A0 B0; define this as Z=0, the other coords are technically unnecessary but x3g requires them so gpx will make some up
G1 Z5 ; move the platform 5mm down
G28 Z F500 ; so we can home z again slowly
M132 X Y Z  ; Recall stored home offsets

G92 E0 ; set current extruder position as 0 so that E15 below makes sense
T{adhesion_extruder_nr}
G1 X112 Y-72 Z0.2 F2400.0 ; move to just on the bed
G1 X-112 Y-72 E15 F1200.000 ; extrude a line of filament along the front edge of the bed
G92 E0 ; set E to 0 again because the slicer's next extrusion is relative to this 0
```

The bizarre thing is that I was previously running my prints through OctoPrint's GPX plugin (using Cura's built-in "Print with OctoPrint" button), and *that* didn't have any problems - prints were correctly centered in the middle of the bed (with the "extrude a line of filament" line correctly going along the front edge) and everything. I'm not sure what OctoPrint did that this plugin isn't doing (or vice versa), but for some reason, when I run the x3g code exported by Cura off of my printer's SD card, that first extruded line goes down the *middle* of the bed, stopping at the *center*. (I tried exporting the G-Code from Cura and converting it by running GPX manually outside the context of the plugin, but saw no change in behavior.)

markwal, 2018-01-16:
Hmmm...  That seems like it should work.  Could you take a 20mm cube and slice it to gcode and the x3g, zip those up and share a link from somewhere like DropBox.  Share a link to this github issue?

stuartpb, 2018-01-16:
Slicing [this](https://www.thingiverse.com/thing:34553) cube from Thingiverse:

[QTI_20mm_cube.zip](https://github.com/Ghostkeeper/X3GWriter/files/1633673/QTI_20mm_cube.zip)

Ghostkeeper, 2018-01-30:
Maybe your printer has the zero point wrong in Cura itself. What happens when you click your machine name in the top right, then "Manage Printers...", then "Machine Settings" and toggle the "Origin at center" checkbox?

Ghostkeeper, 2018-01-30:
Or maybe there is a toggle in GPX to center the g-code?

DavidHaitch, 2018-02-05:
I am experiencing a similar problem with my Replicator Dual clone. It prints in the back right corner. Additionally, my print bed starts the print by going all the way down, all the way up, half way down, and then a quarter of the way up. The end result is that it's printing in midair. 

I tried toggling the Origin At Center checkbox (setting it to True), and it had no effect.

stuartpb, 2018-02-16:
Also, when I tried printing a dual-extrusion print through OctoPrint a couple weeks ago, I got very strange behavior where, on tool change, the tool head moved to the opposite side of the printer (I think it tried to move *more than the entire length of the printer* first, there was some dreadful grinding at one point).

stuartpb, 2018-02-16:
Maybe it would help to compare the GPX output with the code produced by Makerbot Desktop (which produces working output)?

[20mm_cube_makerware.zip](https://github.com/Ghostkeeper/X3GWriter/files/1732901/20mm_cube_makerware.zip)

gunplar, 2018-05-24:
Hmm it seems my printer is from the same company but just different name. It also uses sailfish firmware and set origin at center printbed. Cura X3G plugin does not work properly for me as well. What I do is use ReplicatorG (the version provided by the manufacturer) to translate the x3g file from Cura gcode and it works fine. I read from this post that it seems different X3G translator work differently: https://www.reddit.com/r/3Dprinting/comments/7aqrs1/anyone_got_any_experience_with_x3g_files_and/
If you can find out how to fix this and use Cura's plugin for convenience, please tell me as I'm also needing it.

The origin setting on Cura does not affect anything at all. I just include all those definition at the start of gcodes to be generated and that's all.