Ghostkeeper/X3GWriter #6: X3GWriter fails in Cura 3.2 Beta

DavidHaitch, 2018-01-23:
When attempting to save an X3G file in Cura 3.2.0-BETA, I receive the following message:

    Warning: Could not save to D:\model.x3g: 'Scene' object has no attribute 'gcode_list'. 

Cura then creates an empty X3G file at the location. I don't get this error in Cura 3.1.
I am using version 2.6.0 of X3GWriter, which is what the Cura plugin browser brought me.

Ghostkeeper, 2018-01-28:
Indeed, this is broken due to the multi buildplate feature. I'll take a look and update the plug-in in a way that works for both 3.1 and before, and 3.2.

Ghostkeeper, 2018-01-28:
I'll upload an update tomorrow. Keep an eye on the plug-in browser.

closed 2018-01-28 by Ghostkeeper

usermcusyface, 2018-02-07:
Came here with this problem. Installing the update fixed the issue for me.

RDiazPR, 2018-06-27:
Not working for me in versions 3.2, 3.3, 3.4..... Same error as OP, I had to go back to version 3.1.0 for the X3G plugin to work

Ghostkeeper, 2018-08-08:
Where did you get that plug-in version from that didn't work, @RDiazPR?

commit 0e7139ad (2018-01-28): Get g-code from g-code writer instead of from scene

So if the g-code changes position again we'll update right along. Only if the g-code writer now changes name then we're broken again, but let's assume that this doesn't happen.

Fixes #6.