Ghostkeeper/X3GWriter #11: X3GWriter's output causes CTC printer to behave incorrectly

DavidHaitch, 2018-06-20:
If I use X3GWriter to export from Cura 3.3.1, the printer does the following, in order:
* neglects to heat bed
* heats nozzle normally
* Moves bed upwards very slowly for a few CM, then quickly the rest of the way up
* Moves bed downwards _very_ slowly

I watched it move the bed downwards at a snail's pace for three minutes before giving up

However, if I export from Cura to GCODE, open that GCODE in MatterControl 1.7.5 (another slicer), and export it to X3G using that slicer's GPX integration, the file prints normally.

I've dug into the Cura logs, and into MatterControl's source code, to extract the exact way each is using GPX.
X3GWriter uses GPX 2.6.0, and executes the following: 
gpx.exe -c configFile inputFile outputFile

MatterControl uses GPX 1.3, and executes the following: 
gpx.exe -p -m r2 inputFile outputFile

I've attached the contents of the config file that X3GWriter is using, the GCODE generated by Cura, and both X3GWriter's and MatterControl's X3G outputs. The last three are zipped in X3GWriter_Comparison.zip.
[config.txt](https://github.com/Ghostkeeper/X3GWriter/files/2121233/config.txt)
[X3GWriter_Comparison.zip](https://github.com/Ghostkeeper/X3GWriter/files/2121235/X3GWriter_Comparison.zip)

Ghostkeeper, 2018-07-17:
X3GWriter version [4.]1.1.1 may fix this for you, but only if your printer provides a metadata entry to indicate what flavour of X3G we should write. If you have a machine definition (.def.json file) for your printer, you can add this line to the metadata section:

    "machine_x3g_variant": "r2",

If you don't have a machine definition, you can alternatively also put that metadata in your machine instance. To do this you would go to Help -> Show configuration folder. Find the `machine_instances` folder there and there should be a file with a name that you recognise as your printer. In that file, add these lines:

    [metadata]
    machine_x3g_variant = r2

If there is already a `[metadata]` section, just put the second line under there.

The intent of this is that the designer of the printer definition can set this metadata entry for you and you wouldn't need to worry about it. However with printer definitions being sparse and far between, some manual work is needed.

Ghostkeeper, 2018-11-18:
I'll assume that this was fixed in version 1.1 for you, then.

closed 2018-11-18 by Ghostkeeper