Ghostkeeper/X3GWriter #13: Gpx provided with X3Gwriter doesn't launch

MateuszKrawczuk, 2018-08-17:
I got error "Libwinpthread-1.dll is missing" when I try manually launch gpx.exe

Ghostkeeper, 2018-08-27:
Does the same happen when saving a file as X3G from within Cura? If not, this is not a bug.

Cura provides libwinpthread-1.dll. You should be able to launch gpx.exe from Cura's main directory, like this:

    cd "C:\Program Files\Ultimaker Cura 3.4"
    ..\..\Users\MateuszKrawczuk\AppData\Roaming\cura\3.4\plugins\X3GWriter\X3GWriter\gpx.exe

Alternatively, you could copy libwinpthread-1.dll from Cura's installation folder to wherever gpx.exe is and it should run.

Ghostkeeper, 2018-12-13:
I consider launching gpx.exe from the correct folder a sufficient workaround.

closed 2018-12-13 by Ghostkeeper