Ghostkeeper/X3GWriter #14: X3G USB streaming

cdrose, 2018-09-07:
I found some mention that X3GWriter may now support sending over USB, but I cant find any documentation on how this works or how to configure Cura. Is this currently possible?

Ghostkeeper, 2018-09-08:
Well, basically, X3GWriter since version 1.1 be output to any output device from Cura, not just the local file system. This means that if you put in a USB stick or something and click "Save to Removable Drive" then it would save it.

I think streaming via USB won't work though. There are three things required to make it work:

1. Your printer definition has X3G configured as its preferred output format, instead of g-code. This means that it won't work with Custom FFF Printer and many others. Only for printers that have X3G selected by default when you save to a file. This is because USB printing doesn't allow you to choose the output format. It will only choose the default.
2. The printer has to accept an X3G file being streamed directly. I know that this is common for g-code but X3G is a sort of container format so it might not be accepted in this way by your printer.
3. Cura's normal USB printing plug-in detects USB printers by sending g-code to the connected USB devices. If the printer responds properly to the temperature request g-code command, then Cura should recognise it as a 3D printer. That means that your printer will need at least some support for g-code as well as X3G.

Mainly point 3 is probably a deal breaker. If your printer supports g-code, why would you ever run on X3G?

Ghostkeeper, 2018-09-16:
No, I don't think it will be any different. To get this to work, I figure that it's necessary to implement a USB printing plug-in for Cura that connects up to those printers. Cura's default plug-in won't do it.

closed 2018-09-16 by Ghostkeeper