Ghostkeeper/X3GWriter #16: Configuration documentation - machine_x3g_variant

castaway, 2018-10-05:
I too am having issues with #3 - I've installed the latest plugin release from Github (1.1.3). When writing this now gives me an error: could not save 'NoneType' object has no attribute 'write'

My assumption is that it cant find the "machine_x3g_variant" setting.. and neither can I, where do I edit the configuration for this thing? I've looked in all the Cura (3.4) menus I can find, and in all the files under .local/share/cura (linux) .. no joy..

Any hints please?

TheFirstAxel, 2018-10-09:
I am having this same issue and have not been able to find a way to fix it. And when I try to install the newest version from here it just crashes cura

Ghostkeeper, 2018-10-18:
> When writing this now gives me an error: could not save 'NoneType' object has no attribute 'write'

This was actually a bug in Cura. The SDK version of X3GWriter was outdated, and so Cura thought that it wouldn't interface well with its API and disabled the plug-in. But Cura didn't remove the X3G option from the available file formats, so it still tried to save as X3G and then got a `None` object instead of my plug-in.

I've since released a new version of X3GWriter. Could you update the plug-in in Cura's toolbox and try again?

Ghostkeeper, 2018-10-18:
> My assumption is that it cant find the "machine_x3g_variant" setting.. and neither can I, where do I edit the configuration for this thing? I've looked in all the Cura (3.4) menus I can find, and in all the files under .local/share/cura (linux) .. no joy..
> 
> Any hints please?

While it has no relation to this bug, I'll answer this too. This metadata is located in the machine definition files only. It is not editable through the interface of Cura. You can find these machine definition (*.def.json) files in the `definitions` resource folder in your installation. For instance, look at `C:\Program Files\Cura 3.5\resources\definitions\malyan_m180.def.json`.

TheFirstAxel, 2018-10-18:
Updating through the cura toolbox has fixed this issue for me.

castaway, 2018-10-20:
> While it has no relation to this bug, I'll answer this too. This metadata is located in the machine definition files only. It is not editable through the interface of Cura. You can find these [...]

I have a ~/.local/share/cura/3.4/definitions/ dir, but its empty..

Ghostkeeper, 2018-10-24:
> I have a ~/.local/share/cura/3.4/definitions/ dir, but its empty..

Yes, you could put your own definitions there. But there is also a folder for built-in definitions which will be inside the AppImage.

On Linux, while Cura is running, a disk image will be mounted in your `/tmp` directory that you can explore. In my case of this writing, an image is mounted that contains the definition files in `/tmp/.mount_ydaFEi/usr/bin/resources/definitions`. The AppImage format is an ISO disk image, so if you'd like to edit that you can. Usually it's easier to put a new definition and its extruders in your configuration folders in `~/.local/cura` though, if you just want to test something.

Ghostkeeper, 2018-10-24:
I'm filing this as an enhancement of documentation that needs to be put in the readme. Not during work time though (I'm at work right now).

commit 14df4972 (2021-04-11): Better document how to support additional printers

Fixes #16.
closed 2021-04-11 by Ghostkeeper