Ghostkeeper/X3GWriter #17: Is it possible to edit steps per mm with X3GWriter?

tyiphius, 2018-10-21:
I want to upgrade my stepper drivers on a Sailfish based board for my CTC Bizer. In order to upgrade I have to update the Steps per mm. I know ReplicatorG you can edit an .xml file. I was curious if it was possible to do the same with X3GWriter so it would work with cura as well. (I am by no means professional and I don't know python) I tried opening files with Notepad++ when I came across the X3GWriter.py file. I found a line in that file that reads **parser["x"]["steps_per_mm"] = str(global_stack.getProperty("machine_steps_per_mm_x", "value")) #How many steps of the stepper motor results in 1mm of movement for the print head.** I don't know what format I need or where to place my value to change this. Would I need to change anything else in the file as well? If possible can you give me an example of how the line should look? (You may have to dumb down your explanations for me )
I'm trying to make a guide for others who are limited by these motherboard. Thank you for your time!

Ghostkeeper, 2018-10-24:
Yes, it's possible. You need to modify one of Cura's settings.

The easiest way to do this, in my opinion, is to go into the Toolbox and download the Printer Settings plug-in. This will add all the machine settings that are normally hidden to your list of settings, like the printer size, shape of the build plate, etc. One of these settings will be Steps per Millimeter (X). You can then easily adjust it.

The way that this is meant to be used, is that a printer definition (one of the list of printers that you can choose from when you add a new printer in Cura) will specify the steps per mm in their definition file (.def.json files in the installation folder/resources/definitions) and the user cannot edit them. But if you do want to edit them, the Printer Settings plug-in is there to help you out.

UM2user, 2018-10-31:
Hi there Ghostkeeper,
First of all thanks for this great plugin! It is a great tool to be able to easy export to my private Malyan M180 from Cura directly. The only problem I have is that the Z steps/mm somehow not match the printer. 
I eddited the Json file by hand and also used the printer settings plugin you mentioned (great plugin also) but nothing seem to happen when I alter these settings. The printer moves only a very minimal step down for each layer. could this be a problem of your plugin? is there a way to see the output of your tool? the screen pops up, i see something stated but before I am able to read the first word the screen already closes. I am using the latest version of Cura b.t.w.

Thanks in advance for your reply!
Michel

Ghostkeeper, 2018-12-13:
> The printer moves only a very minimal step down for each layer. could this be a problem of your plugin?

Could be, but I have no real way to debug this without access to your printer :( I suspect it's some interplay between GPX (the g-code to X3G conversion application that this plug-in uses) and your printer. I re-checked the Steps per Millimeter (Z) setting and it seems to work all right from here, but from here I just get a binary file and I can't really see how that would print on your printer.

> is there a way to see the output of your tool?

No, I've been struggling with that as well. There seems to be no application that can load an X3G file and give a preview of it to verify that it is correct. At least, I couldn't find any. I can only really see that the file is not empty and trust on the community to verify it.

> so i suspect that either Cura did not provide the correct Gcode

This can be verified. You could save your print as g-code and load that in a g-code viewer (such as Cura or Repetier, or by manual inspection).

> Also the acceleration settings are not translated which is quite sad, the thing bangs from one side to another which is not that great for smal detail printing.

Yes, this is a known issue. The problem is that the conversion application from g-code to X3G that this plug-in uses does not support acceleration or jerk commands. It outputs a bunch of warnings about this in the log for me. Also a problem is that Jerk is a Marlin-specific concept, and firmware that uses X3G is not based on Marlin. This means that it probably won't be fixed soon. I would expect it to use some sane default acceleration/jerk setting in the firmware though.

UM2user, 2018-12-13:
Thanks for your reply,

In the meantime I found a way to communicate with my M180 and use the code from Cura.
This also proves that Cura does make the right Z steps in Gcode.

I have succesfully used the GPX converter with help from the GPXUI to make things work without command prompt The only thing is the extra step to first save to Gcode and than make the conversion.

The acceleration settings cannot be altered, although i found a setting within the printers firmware to enable or disable acceleration. With this enabled the banging stops :-) so i can use my printer with Cura now, but unafortunatly i cannot make use of your plugin... how can i help to fix it? :-)

Best regards,
Michel

kalniss, 2019-01-14:
Great plugin, it has helped me move almost fully to Cura with my CTC clone, however I can't seem to get the steps per mm to work. I have set up a custom json for my printer. I tried the printer settings plugin. I see my json values through printer settings plugin in Cura, but regardless of changing them in json or in the plugin I see no effect in printer.

I have upgraded my nozzle to E3D Titan so now my machine_steps_per_mm_e are about 4,3 times the original and so far the only way I can use Cura is by setting flow to 430%, which has various adverse effects.

Thank you!

Ghostkeeper, 2020-03-31:
> The acceleration settings cannot be altered, although i found a setting within the printers firmware to enable or disable acceleration. With this enabled the banging stops :-) so i can use my printer with Cura now, but unafortunatly i cannot make use of your plugin... how can i help to fix it? :-)

When you enable Cura's acceleration control, GPX will complain that it doesn't support acceleration commands. So I'm afraid that it's just not supported by the application that converts the g-code to X3G. There is nothing I can really do about this, especially while not having a printer that prints X3G files :(

Ghostkeeper, 2020-03-31:
> I see my json values through printer settings plugin in Cura, but regardless of changing them in json or in the plugin I see no effect in printer.

Can I see that JSON file?

Be aware that if you have the `machine_x3g_variant` metadata set, this plug-in will choose that GPX profile rather than the individual settings such as `machine_steps_per_mm_e`.

Ghostkeeper, 2020-03-31:
I'll close this since the original question has long been answered.

closed 2020-03-31 by Ghostkeeper