Ghostkeeper/X3GWriter #18: Extruders don't heat up?

LukeBonathan, 2018-10-25:
Hi there

I'm now able to export a file using the x3g writer. However the files I export don't engage the extruder heaters. Is there a logical reason for this or might I be missing something?

[cube.x3g.zip](https://github.com/Ghostkeeper/X3GWriter/files/2516693/cube.x3g.zip)

Thanks, 

Luke

Ghostkeeper, 2018-10-27:
I don't know of a good way to decode X3G files to see if it's instructing the printer to heat... :cry: 

There are 4 possible causes I can think of:
* Cura isn't outputting the temperature commands in its g-code.
* This X3GWriter plug-in calls upon GPX to convert g-code to X3G. The plug-in could be calling GPX with the wrong parameters.
* GPX isn't outputting the temperature commands in its X3G even though they are present in the g-code.
* The firmware isn't heating the extruders when reading the X3G file.

The first should be easy to test: Save your slice to g-code instead, and see if there are any M104 or M109 commands in the file. If there are not, then there's two things you need to check in your Machine Settings (under Manage Printers...):
* The g-code flavour must not be set to "ultigcode".
* In the starting g-code, there should either not be any `{material_print_temperature}` or it should be in the last `M109` or `M104` command.

The second can be verified if you have a Cura log of the moment that you're saving the X3G file. X3GWriter puts a log entry of the command that it sends to GPX.

For the third and fourth options, all I can really do is search the internet for bugs. What's your printer, and is the firmware of the printer up to date?

Meganium97, 2021-06-13:
"and no slicer seems to be able to display a print preview or anything."
I am using flashprint and when I import said file, it outputs a "Load file failed" error.
As a side note, anything I export as x3g using this software always will output this same exact error when importing to flashprint.

Ghostkeeper, 2021-06-21:
I've used FlashForge in the past too to visualise it (this post is really quite old) but in its current form (v4.6.0) FlashForge seems to just display a bunch of squares, a bit like what Cura does when the flow is set to 99999%.

Ghostkeeper, 2021-06-28:
Oh, okay. That would probably be good for me to document, then. Thanks.