Ghostkeeper/X3GWriter #19: Start and end scripts for different printers?

moyesboy, 2018-11-07:
I have high hopes to eventually print using cura on my work flashforge creator pro printers where I currently use S3D and I fail to match the quality of what I get of my wanhao at home.
I'm struggling to find start and finish scripts that work for the flashforge creator pro.
My searches for info find lots of people finding the printer begins and then doesnt heat up, few seem to get as far as the print starting.
Is there somewhere to find some start and finish scripts that are known to work for particular printers?
I found some posted for the flashforge creator pro a while ago by ghostkeeper but I couldnt get a print using them either.
I think an issue might be the flashforge is using extruder 0 and extruder 1 and cura is calling them 1 and 2. I'm trying to use extruder 2 so maybe it tries to heat a non existent extruder and stops because it never heats up.

Ghostkeeper, 2018-11-09:
I'm pretty sure that the extruder naming is not an issue here. The output of Cura is still `T0` and `T1`, regardless of what the extruder names are (1 and 2 or Left and Right, doesn't matter).

I don't know where you found any Flashforge script posted by me, but I don't have a Flashforge so if I posted it, I didn't test it! I might have found it somewhere or posted it in name of someone else?

If you got it to work with S3D, maybe you could find the start-up sequence that they are using to start a print and use that?

moyesboy, 2018-11-13:
Ah yes it was not your start and finish codes, you were the next comment here: https://github.com/Ultimaker/Cura/issues/3091

But tracking down start finish codes that work is proving difficult, and the ones I found are rather out of date for current cura version. I try again using info from here: http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html together with info gathered from the comments... but its all very dated.

FireflyNJ, 2018-12-06:
I am too trying to get Cura 3.6 to work with FlashForge Creator Pro.  got the X3GWriter plug in and using the 'Makerbot'  G-code flavor.  I think we just need to figure out the start/end scripts.  I tried others I found posted with no luck but I was trying to get the Left nozzle to work.  But I do know depending on the software some refers nozzles as T0 or T1 and others T1 or T2, so looks like Cura is using 0/1, Left is T1 and right is T0.   I will play with it more and post back here.

JKJameson, 2018-12-08:
I got it working on my Flashforge Creator Pro with the following;

Edit printer config at %AppData%\Roaming\cura\3.6\definition_changes\Flashforge+Creator+Pro_settings.inst.cfg and add this line to the [metadata] section: **machine_x3g_variant = r2**

**Start G-code**
`M73 P0
M107 ; turn off fan
G90 ; absolute coordinates

G130 X118 Y118 A118 B118 ; set stepper motor Vref to defaults
; let the Z stepper vref stay at eeprom level (probably 40)
;
; home and recall eeprom home position
T0 ; home on the right nozzle
G28 X Y Z ; home all axes at homing speed
G92 X0 Y0 Z0 A0 B0 ; set all coords to 0 for now
G1 Z5 F500 ; move Z 5mm away so we can carefully hit the limit switch
G161 Z F100 ; home Z slowly
M132 X Y Z ; recall stored home offsets for XYZ axes
;
; wait for heat up
G1 X110 Y-72 Z30 F3300 ; move to waiting position
; set temperatures and assert Vref
M140 S{print_bed_temperature} T0
M134 T0 ; stabilize bed temperature
M104 S{print_temperature} T0
M133 T0 ; stabilize right extruder temperature
M116 ; wait for temps
;
; purge and wipe
G92 E0 ; set current extruder position as 0 so that E15 below makes sense
G1 X110 Y-70 Z0.2 F2400.0 ; move to just on the bed
G1 X110 Y70 E15 F1200.000 ; extrude a line of filament along the right edge of the bed
G92 E0 ; set E to 0 again because the slicer's next extrusion is relative to this 0`

**End G-code**
`M73 P100 ; end build progress
M140 S0 ; set bed temperature to 0
M104 S0 T0 ; set extruder temperature to 0
M104 S0 T1 ; set other extruder temperature to 0
G1 Z145 F1100 ; send the build plate to the bottom
T0 ; Next job assumes T0 is the current tool
G28 X Y F2500 ; home X and Y axes
M18 ; disable all stepper motors
M70 P5 (Print Complete)
M72 P1 ; Play Ta-Da song`

FireflyNJ, 2018-12-08:
> 
> 
> I got it working on my Flashforge Creator Pro with the following;
> 
> Edit printer config at %AppData%\Roaming\cura\3.6\definition_changes\Flashforge+Creator+Pro_settings.inst.cfg and add this line to the [metadata] section: **machine_x3g_variant = r2**
> 
 
I tried this and at first I though it was not working, but I waited and watched it, was moving so slow I almost couldn't it moving, just hearing the buzz, after almost a minute it suddenly moved and spit out a blob and starting the first layer of the test cube.   I think the priming code is bad.
so maybe other codes I tried were doing the same I just aborted the print before it got to finish the priming. 
. . . Looking into it . . .

JKJameson, 2018-12-08:
So it should actually be machine_x3g_variant = r1d
The code from here in Cura works really well for me: http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html

FireflyNJ, 2018-12-08:
> So it should actually be machine_x3g_variant = r1d
> The code from here in Cura works really well for me: http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html

yea, I tried it, it doesn't like it.  
I am using the latest Cura 3.6, I confirmed I have it set to reprap  and machine_x3g_variant = r1d.

FireflyNJ, 2018-12-08:
check out https://markwal.github.io/GpxUi/commandline.html   ,
I have to try 'machine_x3g_variant = fcp'   when I get home.