Ghostkeeper/X3GWriter #23: What Cura version does this work with

macouno, 2019-01-18:
Hi,

I would love to use the plugin, but it looks like the current version crashes all versions of cura I tried (can't open toolbox).

Can you please tell me what version of Cura I need to be able to enable the plugin (windows 10)

Ghostkeeper, 2019-02-01:
The current version in this repository is made to work with Cura 4.0. Here's the Cura versions vs. the latest X3GWriter versions:

* Cura 4.0 -> v1.1.8
* Cura 3.6 -> v1.1.7
* Cura 3.5 -> v1.1.7
* Cura 3.4 -> v1.1.1
* Cura 3.3 -> v1.1.1
* Cura 3.2 -> v1.1.1
* Cura 3.1 -> v1.0.0
* Cura 3.0 -> v1.0.0
* Cura 2.7 -> v1.0.0
* Anything before 2.7 is not supported.

To get these versions, look into the [releases](https://github.com/Ghostkeeper/X3GWriter/releases) page.

If you can't open the toolbox in Cura, that might be something you'd want to report on the [Cura issues](https://github.com/Ultimaker/Cura/issues) page!

macouno, 2019-02-01:
Thanks for this list. Very useful! I'll be testing on my original replicator next week.

closed 2019-02-01 by macouno