Ghostkeeper/X3GWriter #24: Unable to install on Mac Cura 4.0 Beta

djmuhlestein, 2019-02-01:
Tried latest master direct download, unzip into plugins directory and restart.  Doesn't seem to have recognized the plugin was there.

Also tried dragging the 1.1.18 .curaplugin onto the window to install but Cura says an error that it can't open that type of file.

Ghostkeeper, 2019-02-04:
Indeed, this plug-in is not properly updated for Cura 4.0 yet. Or rather, it was updated for Cura 4.0 but then the API changed so I need to update it again. I'll be sure to do that before the next beta release.

Ghostkeeper, 2019-02-23:
I've since figured out that this functionality is broken in Cura itself, by this commit: https://github.com/Ultimaker/Cura/pull/3663/commits/0f966115e65762149d61009b4968598a5bb33c20

Maybe they'll restore it, but the drag-and-drop of .curaplugin files won't work to install plug-ins in versions 3.6 or 4.0. Sorry! You'll have to use one of the other methods to install the plug-in.

Ghostkeeper, 2019-02-26:
I've added a .curapackage file to the latest release. For Cura 4.0 you'd need to install the package for SDK 6. You can try dragging that in the window.

I'll update the readme later.

djmuhlestein, 2019-02-26:
Confirmed I can drag the new .curaplugin package to the window.  It said it would load on restart.  So... I restarted but I can't find any evidence that it's working.  there is no x3g option in the output file formats.

Even tried adding a new printer that is the Malyan M180 like the readme says to see if it was something in my printer settings causing the plugin to think my printer doesn't need x3g.  Maybe there is some other setting I'm missing to turn it on.

Ghostkeeper, 2019-02-27:
Oh. No, that should work. Maybe you can hand me your log file of Cura? See [Cura's readme](https://github.com/Ultimaker/Cura#logging-issues) on where to find it.

Ghostkeeper, 2019-02-27:
For Cura 4.0, you'll need the SDK 6 version.
It should also be available via the marketplace now.

djmuhlestein, 2019-02-27:
Well it's definitely a 5 vs 6 api version thing.  Here is the message in the log:
Plugin [X3GWriter] with API version [5.0.0] is incompatible with the current API version [6.0.0].

I did download the 6.0 API plugin (at least I thought I did) The thing is, when I installed it from the marketplace, It got the same error.  In both cases I just see the plugin grayed out in the marketplace list of installed plugins.

So then.. I opened up the plugin in my plugins folder and opened up the plugin.json file.  I do see the supported sdk versions are 5 and 6 but the api version is still listed as 5.  So I changed that to 6 and restarted and I can now see the ability to use x3g output files.  I haven't tested things thoroughly with the printer yet but I was able to save an x3g file so at least it seems to be working that far.

Ghostkeeper, 2019-03-03:
I'm not able to reproduce this problem though. I made a fresh install of Cura 4.0 beta 2. Then downloaded the SDK6 version, dropped it onto Cura, restarted Cura and it seems to work fine.

djmuhlestein, 2019-03-05:
Confirmed Beta 2 fixed the issue.  I still had Beta 1 installed and didn't realize there was an update.

closed 2019-03-05 by djmuhlestein