Ghostkeeper/X3GWriter #25: Octoprint plugin

marcinhnat, 2019-03-19:
The plugin doesn’t work, when using fieldofview’s octoprint plugin. The gcode isn’t converted. It saves x3g to SD and to file, but does not work with the octoprint plugin. Is there a solution to this?

Ghostkeeper, 2019-03-20:
It looks like the Octoprint plug-in is [directly calling the g-code writer](https://github.com/fieldOfView/Cura-OctoPrintPlugin/blob/6949f1eb429021f632bc560d494812cf41087c51/OctoPrintOutputDevice.py#L329) to write the file, so the X3GWriter code is never executed then. I'll see what I can do to remedy that.

There is a workaround though, because there is a GPX plug-in for Octoprint (the server-side application) as well: https://plugins.octoprint.org/plugins/gpx/

marcinhnat, 2019-03-20:
Yes, but my issue is that i use AstroPrint which requires directly sending x3g’s to it. The cura plugin is just used to send the file to the Raspberry Pi. I don’t have octoprint on it.

fieldOfView, 2019-03-21:
The OctoPrint Connection plugin is created to cater for OctoPrint. OctoPrint does not even accept x3g files (only gcode and stls), so I don't see how this could work.

Ghostkeeper, 2019-03-21:
All right, so then I suppose that this is working as intended. The OctoPrint is explicitly only sending g-code to it, not any other format, because OctoPrint itself only support g-code.

closed 2019-03-21 by Ghostkeeper