Ghostkeeper/X3GWriter #26: Writing x3g from Cura 4.1/Octoprint 1.3 creates invalid x3g

tomservaux, 2019-07-28:
I recently received a FlashForge Creator Pro for use in printing for a customer contract. We use Cura and Octoprint in our shop and we have had a terrible time with this printer. Some of it was education and we got past that. The problem we finally came up against appears to be with the x3f being generated for the printer. If we print from Cura to Octoprint (using RepRap flavor gcode on both and GPX plugin for Octoprint) the print always fails just above the raft no matter what model we print. One of the errors we get is:

```
2019-07-26 01:13:14,206 - octoprint.util.comm - ERROR - Unexpected error while writing to serial port
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3378, in _do_send_without_checksum
    result = self._serial.write(to_send)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX/gpxprinter.py", line 131, in write
    self._append(gpx.write("%s" % data))
```
We get a different problem when attempting to save to a file instead of using OctoPrint. The file generated is not recognized by the printer and does not show up in the menu). I think it's getting an error related to the previous issue and creating a corrupted file but I can't be sure since I can't decompile x3g binary files.

I've attached an octoprint log file, the 3mf file for my Cura project and the resulting x3g file created when saving to the disk. Any help you can give me would be very very appreciated.

[FCPRO_Cura FlashForge_Test_with_Voids_V2.zip](https://github.com/Ghostkeeper/X3GWriter/files/3439618/FCPRO_Cura.FlashForge_Test_with_Voids_V2.zip)
[octoprint.log](https://github.com/Ghostkeeper/X3GWriter/files/3439614/octoprint.log)

tomservaux, 2019-07-28:
I forgot to mention that I attempted to save this as an x3g using the PgxUi app based on this plugin and had the same result with the file not showing up on the printer when I try to print from the SD card.

Ghostkeeper, 2019-07-29:
When printing via Octoprint, the data never even passes through this plug-in so there is little I can do about that.

When saving to file, the data does go through this plug-in. I also don't have a way to decompile X3G though. No tool online seems to be able to do that and I don't have a printer that reads the file format.

I'll see tonight or tomorrow how I can pass on the error upstream, either to GPX or Octoprint.

Ghostkeeper, 2019-07-30:
When that project file is sliced, GPX puts the following warnings in Cura's log file (through my plug-in):

```
(line 19) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 21) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 22) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 23) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 416827) Semantic warning: M140 cannot select non-existant heated build platform
(line 19) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 21) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 22) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 23) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
current position defined as X:0.00 Y:0.00 Z:15.00 A:0.00 B:0.00
(line 416827) Semantic warning: M140 cannot select non-existant heated build platform
```

Indeed at line 19, 21, 22 and 23 there is a G92. I think it's complaining that there hasn't been a command yet to move the print head so it doesn't know what the current coordinates are. That's kind of the point of the G92 command though: Cura defines the current filament position as 0 at the beginning of the print, so all the rest are relative to that.

At line 416827 there is a command `M140 S0`. This is supposed to set the build plate temperature to 0, but GPX interprets that as if the `S0` refers to the 0th build plate, which I believe is a Makerbot thing, upon which your printer's firmware is based. So here's what I advise you to try:
* In the printer drop-down click on Manage Printers.
* Click on "Machine Settings".
* Under "End g-code", change the second line from `M140 S0` to `M109 S0`.
* Also change the g-code flavour from Marlin to Makerbot.
* Then try a small test print.

Let me know if that works!

tomservaux, 2019-07-30:
I appreciate the help but I ended up returning the printer and buying a different one. Sorry to make you go through the effort. I forgot to update the issue on GH.

Ghostkeeper, 2019-07-30:
All right! I'll close the issue then. Thanks for the report anyway, and maybe that workaround works for other people some day.

closed 2019-07-30 by Ghostkeeper