Ghostkeeper/X3GWriter #27: 1.1.9 has unsupported operand type(s) for *: 'NoneType' and 'int'

djmuhlestein, 2019-08-20:
From Cura.log:

> /Library/Application` Support/cura/4.2/plugins/X3GWriter/X3GWriter/X3GWriter.py", line 195, in write_cfg
> 2019-08-20 08:56:21,261 - ERROR - [Thread-8] UM.Logger.logException [88]:     parser["z"]["max_feedrate"] = str(global_stack.getProperty("max_feedrate_z_override", "value") * 60)
>2019-08-20 08:56:21,265 - ERROR - [Thread-8] UM.Logger.logException [88]: TypeError: unsupported operand type(s) for *: 'NoneType' and 'int'

Haven't found yet where to set max_feedrate_z_override but should the plugin be protecting against a None value as well?

Ghostkeeper, 2019-08-21:
Oops! That setting was removed from Cura in version 4.2.

closed 2019-08-21 by Ghostkeeper

Ghostkeeper, 2019-08-21:
I've made a fix for Cura 4.2. I've published it here on my releases page: https://github.com/Ghostkeeper/X3GWriter/releases/tag/v1.1.10

I've also submitted it to Ultimaker but that needs to be reviewed first and the person who does that is on vacation right now. Could take a week or two.

djmuhlestein, 2019-08-21:
Yep, I saw the commit msg.  Quick work thanks.

commit eb56f220 (2019-08-21): Fix getting max Z feedrate in 4.2

Fixes #27.