Ghostkeeper/X3GWriter #28: Where are the profiles?

ColonelThirtyTwo, 2019-09-03:
The README.md file says that this plugin comes with profiles for various MakerBot printers:

```
X3GWriter supports all parameters of GPX to convert g-code to X3G. Our version of GPX has profiles for the following printers:

    Cupcake Gen3 XYZ
    Cupcake Gen4 XYZ
    Cupcake Pololu XYZ
    Core-XY with heated build plate, single extruder
    Clone R1
    Replicator 1
    Replicator 2
    Replicator 2X
    TOM Mk6
    TOM Mk7
    ZYYX
    FlashForge Creator Pro
```

I don't see these in the source code anywhere, and the Cura plugin in the marketplace only comes with a single profile "MakerBotReplicator" that only has support for one extruder. I'm specifically looking for the Replicator 2X profile.

Where is it? Is this message outdated?

Ghostkeeper, 2019-09-05:
These are not Cura profiles but profiles for GPX, the application that this plug-in uses to convert g-code from Cura into X3G.

In this plug-in's source code there is only a list of known printers here:

https://github.com/Ghostkeeper/X3GWriter/blob/f3066e50160e43759fca24178da3fcd3a5f031b4/X3GWriter.py#L134-L155

The printer that's being used must be defined in the machine definition. One such example is here:

https://github.com/Ultimaker/Cura/blob/e02f0f1480551b5131c94f4741ba36307c70d9eb/resources/definitions/malyan_m180.def.json#L10

The actual profiles are then here: https://github.com/markwal/GPX/blob/master/src/shared/std_machines.h

It's not very obvious to figure out though. Is there something I could improve in this documentation?

ColonelThirtyTwo, 2019-09-07:
Well, when a Cura plugin says it comes with profiles, I expected Cura profiles. It's extremely inconvenient that you don't provide them.

I've tried searching for a cura profile for the replicator 2x with no luck. I've tried making my own profile myself and failed; the startup code I used causes the printer to move out of bounds forever, and I can't find the distance between the two extruder nozzles for the 2X.

Bundling in a set of profiles, just as Cura does with other printers, would make this plugin "just work" when installed.

Ghostkeeper, 2019-09-09:
I cannot provide them, since I have no Replicator 2x (or any of the printers in that list).

That part of the readme is referring to GPX which is bundled with this plug-in. GPX calls them "profiles" so I figured it would be logical to call them that as well.

closed 2020-03-31 by Ghostkeeper