Ghostkeeper/X3GWriter #29: No method to define profile usage?

JesseCake, 2019-10-22:
I realise this is almost been discussed in another issue, but I wasn't able to solve this problem. How are the "profiles" attained by the plugin from Cura? 

I attempted naming the machine profile "Replicator 2" and various other things, but nothing would change the plugin from assuming replicator 1. I ended up having to modify the plugin on my computer in a hacky way to force it to use the Replicator 2 profile as my prints continued to come out the wrong size (different steps per mm) like this:
[](https://community.ultimaker.com/topic/29887-makerbot-replicator-2-and-x3gwriter-plugin-howto/?tab=comments#comment-249038)

I'm using Cura 4.3 in Linux (Ubuntu 18.04).

Any pointers would be great as there are many of us out there still with perfectly working Replicator 2s that now use Cura as Makerware support of both Linux and Replicator 2 has now ended.

Jesse

Ghostkeeper, 2019-10-22:
The plug-in reads metadata from the definition to define which profile it's using. This metadata entry is `machine_x3g_variant`. For instance this printer specifies that it should use the Replicator 1 Dual profile (instead of single):

https://github.com/Ultimaker/Cura/blob/fc0b8185b7df0f223a67f24384e3e6bad663c8a1/resources/definitions/malyan_m180.def.json#L9

If you've made your own printer definition, setting the `machine_x3g_variant` to `r2` should make it use the Replicator 2 profile.

closed 2019-11-04 by Ghostkeeper

JesseCake, 2020-05-25:
Thanks so much - I'm checking it out now and will attempt to add a Replicator2 printer definition to Cura's github so it can come with Cura and save people adding the metadata by hand themselves.