Ghostkeeper/X3GWriter #32: X3GWriter doesn't slice models correctly

dominoboy, 2020-01-09:
Using Cura with the X3GWriter plugin for my Creator Pro but it didn't seem to even attempt to run the file. I used FlashForge to check the file and only the brim appears and no sight of the models. Is this something wrong on my part or is the plugin outdated?

Here's a model in cura and the outputted file opened in FlashForge

[image]
[image]

Ghostkeeper, 2020-01-14:
It's most likely either:
* the configuration in the printer definition doesn't match what Flashprint expects (e.g. steps per mm), or
* some bug in GPX that does the conversion. All this plug-in really does is call on [GPX](https://github.com/markwal/GPX) to convert Cura's g-code to X3G.

At home I'll see what I can do to reproduce your problem (currently at work). Any particular definition file I'll need?

Ghostkeeper, 2020-01-29:
Sorry, this got moved to the background as I was very busy the past few weeks. Back now.

I've tried to reproduce this issue but FlashForge doesn't seem to want to load the X3G files that GPX generates at all. What printer definition file are you using? Can you share them? The .def.json files for both the printer definition and the extruder definitions?

eugr, 2020-02-16:
@dominoboy - try my profile. As far as I know it's the only correct and working Cura profile for FlashForge Creator Pro currently. I've been using it myself for over a month without any issues. I don't save to x3g anymore as I upload gcode to Octoprint, but since it runs GPX to convert gcode anyway, it should work for you. 

https://github.com/eugr/Flashforge-for-Cura

dominoboy, 2020-02-22:
Hello again, sorry for the very long reply, I honestly forgot about using Cura the issue I opened up and continued using FlashPrint for a bit. I've come back now.

I was previously using my own definitions but after using @eugr 's I can finally get a model produced from it, but there's another issue. I used the FlashPrint to double check the file worked properly (I used a benchy as a test model), but it only recognizes the whole model as one line, and the model looks very jagged.
I tried letting my FFCP to run the program but only the bed heats up and not either of the nozzles. I'm guessing this was an error with GPX so I'll try using Octoprint and see if that works

Ghostkeeper, 2020-03-30:
I see what you mean with the jaggedness:
[image]

However I think this is just a result of the visualisation of FlashPrint. They use cuboids with rectangular corners, while Cura visualises lines with triangular endpoints.

If you disable the "Compensate Wall Overlaps" setting in Cura it seems to be less weird at the corners, but I don't know if it's FlashPrint that did this, the X3G representation not taking different line widths well, or something that goes wrong in GPX. I can't say whether or not that works better for the actual print though.

I have no idea what to do about your print not heating up. Are you experiencing the same problem @eugr?

eugr, 2020-03-30:
@Ghostkeeper - no, I haven't experienced this using my profile (nozzles not heating up). Granted, I'm not slicing into x3g, I slice into G-code and upload it to my Octoprint instance that runs GPX to convert it to GCode, but the end result should be the same.

Ghostkeeper, 2020-03-31:
That depends if Octoprint is calling GPX the same way I guess. Maybe my plug-in isn't providing the nozzle temperature correctly if a GPX profile is being used?