Ghostkeeper/X3GWriter #33: 'ascii' codec can't encode character '\u2014' in position 95: ordinal not in range(128)

posttenebre, 2020-02-17:
Dear Developer Ghostkeeper,

Thank you for writing the X3GWriter plugin for cura. I was wondering if you could point me in the right direction for resolving the issue when writing the file to .x3g.

When saving a file I get the warning that it could not save the file because of:

'ascii' codec can't encode character '\u2014' in position 95: ordinal not in range(128)

Cura 4.4.1
X3GWrite 1.1.11

Thanks in advance

Ghostkeeper, 2020-02-17:
Could you share the Cura project file (under File -> Save...) so that I can try and reproduce the problem?

Where are you seeing that error? Is it in the Cura log?

posttenebre, 2020-02-17:
Thanks for getting back to me so quickly.

The message appears at the bottom centre of the screen like all other messages like when saving. I'm just trying to do a quick test print of the pruse bolt and nut.

I uninstalled the plugin and reinstalled it. I also upgraded to Cura 2.5 beta. I still get the same message.

Could it be python on my computer?

https://drive.switch.ch/index.php/s/cTPXph6MxGAyRA5

Ghostkeeper, 2020-02-25:
The Python version on your computer shouldn't affect it, since Cura comes with its own version of Python (unless you're not using one of our distributions; on Linux there are some alternative distributions that do use your system's Python version).

I can reproduce the issue with that project file. I'll see if I can make a fix for it that filters out those non-ASCII characters.

Here is also a workaround: Just don't use non-ASCII characters in the start g-code or end g-code. Here is a new start g-code for you where I replaced them with their ASCII equivalents:

```
; - start of START GCODE -
M73 P0 (enable build progress)
M92 X88.8 Y88.8 Z400 E101 ; sets steps per mm for Rep2
G90 (set positioning to absolute)
(**** begin homing ****)
G162 X Y F4000 (home XY axes maximum)
G161 Z F3500 (home Z axis minimum)
G92 Z-5 (set Z to -5)
G1 Z0.0 (move Z to "0")
G161 Z F100 (home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
(**** end homing ****)
G92 X147 Y66 Z5
G1 X105 Y-60 Z10 F4000.0 (move to waiting position)
G130 X0 Y0 A0 B0 (Set Stepper motor Vref to lower value while heating)
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
G0 X105 Y-60 (Position Nozzle)
G0 Z0.6 (Position Height)
```

```
; - start of END GCODE -
G92 Z0
G1 Z10 F400
M18
M104 S0 T0
M73 P100 (end build progress)
G162 X Y F3000
M18
```

After replacing those, it seems to work correctly for me.

posttenebre, 2020-02-25:
It worked. Thank you so much for taking the time to look into this.

Ghostkeeper, 2020-03-30:
I've made the plug-in filter out any characters not in the ordinal range now, so that GPX won't crash on unusual start or end g-code any more.

commit 15d62970 (2020-03-30): Filter out any non-ASCII-7 characters

GPX crashes on those. They shouldn't be necessary in the g-code anyway.

Fixes #33.
closed 2020-03-30 by Ghostkeeper