Ghostkeeper/X3GWriter #34: cura 4.5 doesn

asfcarvalho, 2020-03-27:


asfcarvalho, 2020-03-27:
When I try to use the X3GWriter in CURA 4.5 the printer doesn't work, just in CURA 4.2.1

Ghostkeeper, 2020-03-31:
I'm unable to reproduce this problem. The plug-in seems to load fine in Cura 4.5. As you can see I saved a .x3g file here:
[image]

Are you sure you've updated the plug-in? Cura 4.4 and 4.5 are supported starting from version 1.1.11.

alyssafriesen, 2020-04-13:
I am having an issue with this as well. Cura 4.5 and XG3Writer 1.1.11. Everything appears to work, but the file it creates is empty. Log file is here:

`
2020-04-13 12:51:42,024 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [129]: Writing to [/Users/alyssafriesen/Downloads/FCPROLE_Test_Cube.x3g]...
2020-04-13 12:51:42,078 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [150]: Writing to Local File /Users/alyssafriesen/Downloads/FCPROLE_Test_Cube.x3g in binary mode
2020-04-13 12:51:42,145 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [6c3cae4515b74400a3412454582d9ae8] started
2020-04-13 12:51:42,188 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-13 12:51:42,349 - DEBUG - [Thread-1] X3GWriter.X3GWriter.write [50]: Using configured machine: fcp (FlashForge Creator Pro)
2020-04-13 12:51:42,360 - DEBUG - [Thread-1] X3GWriter.X3GWriter.gpx_executable [109]: GPX executable: /Users/alyssafriesen/Library/Application Support/cura/4.5/plugins/X3GWriter/X3GWriter/gpx_macos
2020-04-13 12:51:42,361 - DEBUG - [Thread-1] X3GWriter.X3GWriter.gpx_command [130]: GPX command: /Users/alyssafriesen/Library/Application Support/cura/4.5/plugins/X3GWriter/X3GWriter/gpx_macos -g -p -m fcp /var/folders/l8/9n01689j3417c5htpz8s2_nm0000gp/T/tmp6zku32bx /var/folders/l8/9n01689j3417c5htpz8s2_nm0000gp/T/tmp6lgqr4tk
2020-04-13 12:51:42,460 - DEBUG - [Thread-1] X3GWriter.X3GWriter.write [62]: (b'', b'dyld: Library not loaded: /usr/local/lib/gcc/5/libgcc_s.1.dylib\n  Referenced from: /Users/alyssafriesen/Library/Application Support/cura/4.5/plugins/X3GWriter/X3GWriter/gpx_macos\n  Reason: image not found\n')
2020-04-13 12:51:42,485 - DEBUG - [Thread-1] UM.FileHandler.WriteFileJob.run [76]: Writing file took 0.3732318878173828 seconds
2020-04-13 12:51:42,708 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [6c3cae4515b74400a3412454582d9ae8] finished.
2020-04-13 12:51:42,713 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-13 12:51:42,714 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully
`

Digisembler, 2020-05-28:
Same sort of thing here (ish). i have version 1.1.11 on cura 4.5. slices fine, saves fine. When i go to run the file on my makerbot replicator 2x, that's where i'm having trouble...
[image]
[image]
screen images show that the file name is some sort of temp name, the bed warms up, but never clicks over to heat the nozzle. 2nd image shows (under monitoring window) that it's waiting on homing.  not sure what that means, or if i can figure out how to edit that or what to do about it. (since x3g is binary... or unreadable either way).  Any suggestions? thanks.

Digisembler, 2020-05-28:
to add to this - it zeros out, but then doesn't move the bed down and move the heads over to the heat extruder spot (forward left corner, opposite the 0,0,0 spot) (whatever that command is). so, it just hangs. thanks. (old makerbot software generally works ok, but there isn't a horizontal expansion setting in the makerbot s/w, so, getting dimensionally accurate parts is tough without a lot of sanding) (the main reason i want to use cura, other than cura is just way more awesomer...

Ghostkeeper, 2020-06-02:
> screen images show that the file name is some sort of temp name

That's probably because this plug-in writes it to a temporary file, because GPX requires a file name to write to (can't write to a stream) but Cura needs it as a stream. So it creates a temporary file, lets GPX write to that file, and then reads that file back in for Cura.

I don't know what would make your printer firmware wait for homing. Perhaps the steps per mm are not configured correctly and it's homing extremely slowly or something? Or perhaps it thinks that the endstops are on the other side? I think that this is caused by having the wrong `machine_x3g_variant` metadata entry in your printer definition, or wrong machine settings. Which definition are you using?