Ghostkeeper/X3GWriter #35: Output x3g file is zero bytes

BenBergman, 2020-07-09:
I am running Cura 4.6.1 and X3GWriter from the marketplace (1.1.11). I have used the MakerBot Replicator 2 "definitions" and "extruders" files from https://github.com/rpavlik/cura-rep2x. Saving the sliced gcode correctly defaults to x3g, but the resulting x3g file is completely empty. If I instead ask it to save to gcode and use gpx included in x3g to manually convert the gcode to x3g, I get a properly formatted x3g file.

Since the gcode is fine and the manual conversion to x3g is fine, it makes me think there is an issue with X3GWriter, but I would happy to be told there is an error in the config I need to adjust. :)

Ghostkeeper, 2020-07-13:
Strange. Indeed the problem is probably in this plug-in then. Basically all that this plug-in really does is save the g-code to a temporary file and then call the GPX executable on it.

Do you have a log file from Cura for me, after just trying that? See https://github.com/Ultimaker/Cura#logging-issues

BenBergman, 2020-07-13:
Thanks for the link to the logging info. I've posted my cura.log [here](https://gist.github.com/BenBergman/ddfd72eb754dbef406f5f00d667c510e).

The relevant lines seem to be here:

```
2020-07-13 14:14:49,923 - DEBUG - [Thread-5] X3GWriter.X3GWriter.gpx_executable [109]: GPX executable: /Users/benjamin.bergman/Library/Application Support/cura/4.6/plugins/X3GWriter/X3GWriter/gpx_macos
2020-07-13 14:14:49,924 - DEBUG - [Thread-5] X3GWriter.X3GWriter.gpx_command [130]: GPX command: /Users/benjamin.bergman/Library/Application Support/cura/4.6/plugins/X3GWriter/X3GWriter/gpx_macos -g -p -m r2x /var/folders/xy/ld5r5kd94077l898chsd2kbsqtdmq0/T/tmpmclnhj91 /var/folders/xy/ld5r5kd94077l898chsd2kbsqtdmq0/T/tmpf0aybcp9
2020-07-13 14:14:50,164 - DEBUG - [Thread-5] X3GWriter.X3GWriter.write [62]: (b'', b'dyld: Library not loaded: /usr/local/lib/gcc/5/libgcc_s.1.dylib\n  Referenced from: /Users/benjamin.bergman/Library/Application Support/cura/4.6/plugins/X3GWriter/X3GWriter/gpx_macos\n  Reason: image not found\n')
```

Maybe this is another example of macOS Catalina doing dumb things with permissions or something? I feel like I've had an issue in the past with some other program that worked fine run from the CLI but not a GUI because iTerm had special permissions that the GUI program didn't.

BenBergman, 2020-07-13:
Looks like it is related to #30 . Certainly weird that the gpx bin works fine from the CLI and not from cura, but at least I'm getting files generated now.

Ghostkeeper, 2020-07-15:
Indeed, I'll close this as a duplicate then. Thanks for the investigation. We'll continue the discussion there.

closed 2020-07-15 by Ghostkeeper