Ghostkeeper/X3GWriter #36: Installation of plug-in

delidel97, 2020-07-31:
Can someone please advise how to actually install these files so they work on Cura; I've downloaded them w=however have no idea how to install. Please help.

I am trying to combat this error on Cura for reference

Thanks! 
[image]

Ghostkeeper, 2020-07-31:
The easiest way is to go into the Marketplace:
[image]

Then find the X3GWriter down in the list. It's alphabetical so it'll be near the bottom:
[image]

In that page you'll click on the big blue "Install" button:
[image]

Then restart Cura, and it should be installed.

Ghostkeeper, 2020-07-31:
Note that this plug-in doesn't allow you to *open* X3G files. It just allows you to *save* them. I don't know when you were getting that error?

delidel97, 2020-07-31:
Thank you so much!!! I am so grateful for your help

closed 2020-07-31 by Ghostkeeper

delidel97, 2020-07-31:
I was receiving this error when trying to save my sliced files to my USB
before I installed the plug in. Now that I have installed the plug in I was
able to save the sliced file to my drive so I’m guessing this means it has
worked and I can print this file