Ghostkeeper/X3GWriter #37: Windows 10; X3GWriter making empty file

Danmchales, 2020-08-24:
Hello, I'm trying to use Cura with my Flashforge creator pro, but cura only produces a file with no data. Is there a way to fix this?

Ghostkeeper, 2020-08-26:
I don't know. Could you show me Cura's log file?

Danmchales, 2020-08-26:
[Curalog.txt](https://github.com/Ghostkeeper/X3GWriter/files/5130377/Curalog.txt)

Sure, thank you for your help!

mwu1020, 2020-08-27:
Hello Ghostkeeper, I have the same problem like Danmchales.
I can create the x3g file but it is empty. I'm using MacOS 10.15.6 with Cura 4.6/4.7 3d Printer is an Makerbot Replicator 2
log file attached.

Very thanks for your help
Martin

[Curalog.txt](https://github.com/Ghostkeeper/X3GWriter/files/5138483/Curalog.txt)

Ghostkeeper, 2020-08-28:
@mwu1020 there is a similar report for MacOS here, with a lot of discussion: https://github.com/Ghostkeeper/X3GWriter/issues/30

The problem with MacOS seems to be that the GPX binary I'm using is linked to a libc distributable which is not always present on your computer. That's different from what's happening on Windows, so I'd like to keep the issues separate.

Ghostkeeper, 2020-08-28:
Strange, it looks normal from the log file that Danmchales posted. Writing the file took 0.26s, the GPX command gives no output in its error streams. All seems correct to me. Very strange.

You could look if there is any mention of "gpx.exe" in the blocklist of any firewall you may be using. Never heard of the firewall blocking that though, but it could be.

Danmchales, 2020-08-28:
gpx.exe doesn't appear in my firewall, anything else I can try?

swyphcosmo, 2020-09-06:
I had this happen to me too after an upgrade to 4.7. When I opened it, I selected firewall settings to allow comms on Private and Public networks. I tried downloading 4.6.1 (which I uninstalled when I upgraded) and 4.6.2, both had the same issue even though each had their own copies of the X3GWriter plugin. 

I rebooted and removed cura from the firewall settings. When I opened cura this time, I kept the default firewall settings (only Public networks), and poof! It worked!

Hope this works for others too!

kleinwareio, 2020-12-21:
Also having empty `.x3g` file output problem. Win 10, Cura 4.8.0, X3GWriter 1.1.11.

Running Cura as admin (as suggested in another, closed issue) creates `.x3g` files as expected.

charisis, 2021-02-06:
The problem, at least in my case, was that the gpx executable was linked against `libwinpthread-1.dll` which doesn't exist in windows. I believe this comes with some mingw package.

I was able to work around the problem by downloading a fresh (I believe statically linked) binary from the [gpx project](https://github.com/markwal/GPX/releases) and dropping it into the plugin directory.

Maybe you'll want to check the `returncode` of the pipe subprocess after completion to catch abnormal termination cases.

Ghostkeeper, 2021-04-10:
I'm working on a new release of the X3GWriter with updated GPX version and rebuilt binary, as an attempt to fix this. I have a preview here, which you can drop in your Cura window to install. Would any of you like to try?
[X3GWriter7.0.0.zip](https://github.com/Ghostkeeper/X3GWriter/files/6290920/X3GWriter7.0.0.zip)
The file refers to SDK 7.0.0, which should work with Cura 4.4 and newer. (The release would actually be versioned 1.1.12.)

To install this version: Drop the .curapackage file onto the Cura window, then restart Cura.

To uninstall it (if unsuccessful or whatever): Go to the Marketplace in Cura and click on the "Installed" tab. Then search for the X3GWriter and press the "Uninstall" button.