Ghostkeeper/X3GWriter #38: x3g export not respecting acceleration settings

gadg3ts, 2020-09-02:
Hello,
I have a Flashforge Creator Pro that I'm using with Cura & the X3GWriter plugin.
It's also set up with the "Flashforge-for-Cura" profile that is somewhere else on github.
If I export a gcode file for use the OctoPrint + its GPX plugin, that works fine.
However, if I export an x3g file from Cura with the same settings for use in the machine itself, when I try printing the files directly from the SD card, it appears to not respect the acceleration settings with the head speeding around the platform, so I'm getting shifted layers when the toolhead changes.
This seems to happen in 2/3 of the attempts, so it's a bit of a problem.
How can I figure out where the problem is?
I've attached one of the exported files and what the broken result looks like
[FCPRO_C-MajCCv3.zip](https://github.com/Ghostkeeper/X3GWriter/files/5163087/FCPRO_C-MajCCv3.zip)
[image]

Thanks,
Sean

Ghostkeeper, 2020-09-03:
Indeed I'm using [this version of GPX](https://github.com/markwal/GPX) which is as far as I know the most maintained one. When saving the file, that version outputs this to the stderr channel:

    Syntax warning: unsupported mcode command 'M204'

`M204` is the g-code command to change acceleration.

If they got it to work in Octoprint, maybe they are maybe using some pre-processing to change the g-code format to something that GPX does understand.

Ghostkeeper, 2021-04-11:
It seems that the Octoprint plug-in is [this repository](https://github.com/markwal/OctoPrint-GPX). It doesn't do anything extra to support acceleration. It simply calls GPX in the same way as me. I don't know how there could be a difference, unless the GPX machine configuration isn't correct in the Cura settings of that "Flashforge for Cura" profile you found.

Ghostkeeper, 2021-04-11:
The GPX application only supports [enabling or disabling acceleration](https://github.com/markwal/GPX/blob/f22b938e773f4accf9138d8723a4ecfbdefdd2b2/src/gpx/gpx.c#L5759) completely, with the `M320` and `M321` commands. Disabling acceleration would, as far as I can see, effectively set the acceleration to infinity. That's not something you'd want to ever use. Perhaps it would be good to put an `M320` command in your start g-code though to enable acceleration in case it somehow got disabled.

Configuring the printer to print well is out of scope of this plug-in. You'll have to write your own start g-codes.

I'll close this since it is an issue upstream in the GPX program. You can report the issue [here](https://github.com/markwal/GPX/issues), although it is a known shortcoming there.

closed 2021-04-11 by Ghostkeeper