Ghostkeeper/X3GWriter #39: x3g filenames longer than 31 chars not available on FFCP

jrwwallis, 2020-09-15:
I think this is more of a limitation with Creator Pro firmware, but it would be nice if this plugin could work around it.

For a given input .stl, the following .x3g is written to flash:
FCPRO_.x3g

This is fine if the filename length is 31 chars or less, but if >=32, then the file simply doesn't show up on the "Print from SD" menu on the FFCP. For example, if I have the two .x3g files on my SD card:

31.000000000000000000000000.x3g
32.0000000000000000000000000.x3g

Only the 1st shows up in the file chooser menu.

A convenient workaround for this would be for the plugin to truncate .x3g filenames longer than 31 chars to 31 when writing to SD card.

Ghostkeeper, 2020-09-17:
I'm not opposed to Cura having to work with limitations of the printer. This plug-in knows when it's exporting for FlashForge Creator Pro, so it could adjust the file name.

Ghostkeeper, 2021-04-10:
Correction: Sorry, but this plug-in is unable to adjust the file name. It knows indeed when you're using the Flashforge Creator Pro machine. However this plug-in is only an output format, and the way that output file formats are used in Cura is independent of where the output gets written; be it on disk (the user gets to choose a name), on USB stick (the name is automatically defined by Cura), sent via wifi or USB cable (there is no file name at all).

All that I can control from this plug-in is what gets written in the I/O stream. Not where this I/O stream is written to. So I'm afraid that there is nothing I can control from within this plug-in to prevent this mistake. Sorry!

closed 2021-04-10 by Ghostkeeper

Meganium97, 2021-06-13:
Can confirm, 31 characters is the hard-set limit, as the FFCP and I assume its counterparts all cannot read names above this limit. Flashprint even tells you that files that exceed this limit cannot be exported due to it exceeding the file read limit (30 bytes, or 31 characters)