Ghostkeeper/X3GWriter #41: Changing steps per mm with X3GWriter, how?

muffin2, 2021-01-21:
I can't figure out how to change the steps per mm for x/y axes for my Replicator 2 (with Sailfish 7.7) within Cura 4.8.0 and X3GWriter plugin.
I changed the x3g_variant to "r2" in the MakerBotReplicator.global.cfg in the cura machine_instance folder.

According to the issues #17  and #31 and your setting guide i understand that i can change the steps/mm with the printer setting plugin. And as far as i understood, those values are then taken into account from X3GWriter to calcuate the x3g-Code out of the gcode. But no matter what values i enter (e.g 70 steps/mm instead of 88 as default value of Makerbot), it has no effect on the print.
Moreover, it is not possible for me to enter floating values.
I have no idea, what could be wrong.. Does an M92 command in the Start-Gcode has an influence to X3GWriter? So would this be an option?

muffin2, 2021-01-22:
i compared now the sliced gcode and x3g files with the same settings but the steps per mm.
In the gcode the different machine_steps_per_mm_# values are stated at the end (after the end Gcode) in the ;SETTING_3 section, behind "extruder_quality": [" .... [values]... \\nmachine_steps_per_mm_x = xx "]   in a row with other parameters.
But the both corresponding x3g are exactly the same; except the tmp file name in the x3g code.

For me it looks like, that the machine_steps_per_mm_# parameters are ignored or not correctly interpreted by gpx. or that the parametername is not correct in the gcode and thus ignored by gpx.

Ghostkeeper, 2021-01-29:
The steps per mm is not used by CuraEngine at all so indeed the g-code should be exactly the same.

> I changed the x3g_variant to "r2" in the MakerBotReplicator.global.cfg in the cura machine_instance folder.

If the x3g_variant is set, it'll use that pre-determined config in GPX. If not, it'll use the Cura settings. So indeed, setting the steps per mm wouldn't have any effect then. It'll use the settings in the GPX profile called "r2", which are these:

https://github.com/markwal/GPX/blob/28f346005ee37117b1f042dd66570d108726a02f/src/shared/std_machines.h#L186-L201

Sorry if this was unclear. Cura always has a value for every setting, so having Cura's settings override the GPX settings wouldn't make any sense; you're just not using the GPX profiles then.

closed 2021-01-29 by Ghostkeeper