Ghostkeeper/X3GWriter #42: Couldn't save file: "charmap" codec can't encode character '\u200b'

iBNB-Loussa, 2021-01-28:
Dear Ghostkeeper,

first off, great thanks for the X3GWriter plugin. I was trying to save a file under Cura 4.8.0 for the QidiTech printer using your plugin but unfortunately, I couldn't do so due to an issue with ASCII characters somewhere. 

Please find attached a screenshot for cura`s interface with the error message and the printer parameters. 

Is there any cure for that problem ? 

Thanks

[image]

!

Ghostkeeper, 2021-02-04:
Hi!

Unicode character 200B is a [zero-width space](http://www.unicode-symbol.com/u/200B.html). It's essentially point at which words are allowed to wrap, but it's completely invisible (as opposed to a normal space). That's why the start g-code looks normal. But I bet that there's a zero-width space in there somewhere. Perhaps if you copied that start g-code from some website the zero-width space came along.

As a workaround, I suggest that you go through the start g-code with the arrow keys and see if there's any point at which the cursor doesn't advance when you press the right arrow key. That's where the zero-width space will be.

As a fix, I think this plug-in should filter out any characters that it can't send to the GPX converter application.

Ghostkeeper, 2021-04-10:
This is fixed by https://github.com/Ghostkeeper/X3GWriter/commit/15d629706af1232e58a46b8f18dacca9d2941980. Thanks for the report!

closed 2021-04-10 by Ghostkeeper