Ghostkeeper/X3GWriter #47: Error on saving file in Debian subsystem on Chrome OS

GFBsoul, 2021-11-26:
Saving an X3G file fails on Chrome OS, running the Debian subsystem.

To recreate: Install Cura via apt and install X3G writer. Slice and save. Error appears when saving. File is created but contains zero bytes.

Debug from terminal:
`19:37:06.634: gtk_file_chooser_default_set_current_name: assertion 'impl->action == GTK_FILE_CHOOSER_ACTION_SAVE || impl->action == GTK_FILE_CHOOSER_ACTION_CREATE_FOLDER' failed
2021-11-26 19:37:09,283 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [130]: Writing to [/home/gustavbrogren/ZA_back_cover.x3g]...
2021-11-26 19:37:09,285 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [114]: 'info/send_slice_info' is turned off.
2021-11-26 19:37:09,287 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [151]: Writing to Local File /home/gustavbrogren/ZA_back_cover.x3g in binary mode
2021-11-26 19:37:09,514 - DEBUG - [Thread-6] X3GWriter.X3GWriter.write [62]: Using configured machine: z (ZYYX - single extruder)
2021-11-26 19:37:09,516 - DEBUG - [Thread-6] X3GWriter.X3GWriter.gpx_executable [133]: GPX executable: /home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx
2021-11-26 19:37:09,519 - DEBUG - [Thread-6] X3GWriter.X3GWriter.gpx_command [164]: GPX command: /home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx -g -p -m z /tmp/tmpbn60ullz /tmp/tmp49g_ojl1
2021-11-26 19:37:09,580 - ERROR - [Thread-6] X3GWriter.X3GWriter.write [76]: System call to X3G converter application failed: [Errno 8] Exec format error: '/home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx'
2021-11-26 19:37:09,600 - DEBUG - [Thread-6] UM.FileHandler.WriteFileJob.run [80]: Writing file took 0.24532032012939453 seconds
2021-11-26 19:37:09,650 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/MessageStack.qml:75: TypeError: Cannot read property 'horizontalCenter' of null
`

OS: Chrome OS 96.0.4664.57
Debian version: Debian 11 Bullseye
Cura version: 4.8-4
X3G Writer version 1.1.12

Ghostkeeper, 2021-12-06:
Note that you're not running mainline Cura. It looks like the Creality distribution, going by the version numbering.

I guess the key here is this line of the error: `System call to X3G converter application failed: [Errno 8] Exec format error: '/home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx'`

That indicates that the GPX application is not built to run on your system. Which could be; Cura doesn't normally target ChromeOS. The Debian subsystem should execute it properly, but apparently it's not the same as a real Debian system then. I'm running Ubuntu myself.

There's something we can try to narrow the issue down further, if you're willing to help. We could run the gpx application from a terminal to see if we get a similar effect. Something like this:

```
~$ /home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx
```

If the program executes correctly, you'd see this:

```
Command line error: provide an input file or enable standard I/O

GPX 2.6.8
Copyright (c) 2013 WHPThomas, All rights reserved.
Additional changes Copyright (c) 2014, 2015 DNewman, MWalker
All rights reserved.

For usage information: gpx -?
```

If it's not executing correctly, we'll have to raise this upstream to [GPX](https://github.com/markwal/GPX). And most likely we'd have to build something in this plug-in too to detect ChromeOS and select the correct binary, if it needs to be a different binary.

If it is, we can further narrow it down by trying to call it from Python, just like what this plug-in does:
```
~$ python3
>>> import subprocess
>>> subprocess.Popen("/home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx")
```

If it's working correctly, you'd get a similar output again, intermingled with Python's interactive terminal:

```
<Popen: returncode: None args: '/home/trin/Gedeeld/Projects/X3GWriter/gpx'>
>>> Command line error: provide an input file or enable standard I/O

GPX 2.6.8
Copyright (c) 2013 WHPThomas, All rights reserved.
Additional changes Copyright (c) 2014, 2015 DNewman, MWalker
All rights reserved.

For usage information: gpx -?
```

If not, you'd get an error similar to what you saw in the Cura log. If it's not working correctly this way, this is really down to the subtleties of calling applications from Python on your system and I'm not really sure what to do about it, without getting a Chromebook myself (which I'd refuse even if it were offered to me freely).

GFBsoul, 2021-12-07:
`gustavbrogren@penguin:~$ .local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx 
-bash: .local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx: cannot execute binary file: Exec format error`

GFBsoul, 2021-12-07:
Ooops, didn't mean to close the issue. Github permissions doesn't let me reopen it since I'm not a contributor.
Anyway, I would gladly help out as much as I can.

I seem to be running mainline Cura. It is installed from the Debian repos. Screenshot with about page: https://ibb.co/vmRrwmc
Just to be clear about the system. This is the linux virtual machine (or subsystem if you would like to call it that) in Chrome OS. It is a kind of stripped down version that integrates somewhat into Chrome OS.

Output from running the gpx file directly:
`gustavbrogren@penguin:~$ .local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx 
-bash: .local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx: cannot execute binary file: Exec format error`

Output from running it in python:
`gustavbrogren@penguin:~$ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import subprocess
>>> subprocess.Popen("/home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1823, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/gustavbrogren/.local/share/cura/4.8/plugins/X3GWriter/X3GWriter/gpx'`

I see that I'm running Python 3.9. It is quite new. What version is X3GWriter written for?
(The Chromebook isn't too bad. It is crippled, but the Linux VM opens up for some more applications. Most of all, it has got battery life for more than a day of work, which is really nice. Anyhow, the Chromebook discussion isn't the topic of this thread).

Ghostkeeper, 2021-12-07:
No beef with people using Chromebooks, just not for myself :stuck_out_tongue: This is indeed not the right place for it.

> I seem to be running mainline Cura. It is installed from the Debian repos.

Ah, okay. Ultimaker doesn't support the Cura version from the Debian repositories, only the AppImage they distribute. But it should be the same. Anyhow I don't think that's the issue here. It's the GPX binary itself, not the environment, since you can't run the application even directly from the command line.

> What version is X3GWriter written for?

It was written during the release of Cura 15.06 which was when Cura was using Python 3.4. But it's meant to work with any Python version. It calls the GPX application as a subprocess, which works the same way in all recent Python versions.

Since you're seeing the same issue when executing GPX via the command line, it appears that the issue is simply that the executable file is in a format that your computer can't understand. The executable file `gpx` is taken from the [2.6.8 release of that application](https://github.com/markwal/GPX/releases/tag/2.6.8). It is binary identical, which can be verified to confirm this plug-in's security.

So I think this will have to be reported on the [GPX issue tracker](https://github.com/markwal/GPX/issues). I don't expect Markwal to be able to spend a lot of time on it, but he's pretty clever with this sort of compilation issues. I don't know if he can fix it or if you'd have to wait on someone from the community to pick it up. It's not something that I can really do anything about, I'm afraid.

closed 2022-01-12 by Ghostkeeper