Ghostkeeper/X3GWriter #49: X3GWriter for Cura 5.0

Alejandroski33, 2022-05-24:
Update X3GWriter for a compatible version of Cura 5.0

ariacorrente, 2022-05-28:
I tried to force the compatibility of this plugin with the 5.0 version of Cura and changed the `plugin.js` file in the installation directory like this:

https://gist.github.com/ariacorrente/cef00ab1950c6240feb60d427afb0b51

The changed lines are 6 and 7.
I tested some small prints on a Flashforge Creator Pro and it worked fine.

WARNING: My knowledge about Cura, Cura plugins or GCode is zero. I have no idea if the changes are safes or if the plugins needs something else to be up to date with the new API. **If someone wants to force the compatibility before the developer will release an updated version i strongly recommend to watch closely the print and go SLOW.**

I wanted to test the new variable width lines feature but after printing I realized that I am not even sure if this plugin requires some change to support the new feature. For now I thank the developer for his work and add myself to the list of users who would appreciate an update.

Ghostkeeper, 2022-06-04:
It was indeed just the SDK version increment in the JSON. Everything seems to be working fine still.

The new release is published here: https://github.com/Ghostkeeper/X3GWriter/releases/tag/v1.1.13 It's also submitted to the Marketplace, so once Ultimaker reviews it it'll be easy to install from there too.

closed 2022-06-04 by Ghostkeeper