Ghostkeeper/X3GWriter #50: Hello teacher! Can you help me change the GCX output to Gcode? Thank you very much!

a5421306, 2022-11-26:
# Copyright (c) 2015 Ultimaker B.V.
# Cura is released under the terms of the AGPLv3 or higher.

from . import GCxWriter

from UM.i18n import i18nCatalog
catalog = i18nCatalog("cura")

def getMetaData():
    return {

        "mesh_writer": {
            "output": [{
                "extension": "gcx",
                "description": catalog.i18nc("@item:inlistbox", "Encoded gcode File"),
                "mime_type": "text/x-gcx",
                "mode": GCxWriter.GCxWriter.OutputMode.TextMode
            }]
        }

    }

def register(app):
    return { "mesh_writer": GCxWriter.GCxWriter() }

Ghostkeeper, 2022-11-28:
Looks good to me as an `__init__.py` file. Of course you'd need to implement that `GCxWriter` class then.

However this is an issue queue for X3GWriter, a different plug-in. I suggest that you look into [the example plug-ins distributed by Ultimaker](https://github.com/Ultimaker/UraniumExampleFileReaderPlugin), or ask on the [Ultimaker forum](https://community.ultimaker.com/forum/108-ultimaker-cura/) instead.

closed 2022-11-28 by Ghostkeeper

a5421306, 2022-11-29:
Dear teacher, can you teach me to delete this plug-in so that it can output gcode format normally？Another pyc file is encrypted and cannot be decompiled！

Ghostkeeper, 2022-11-29:
Cura comes with a built-in plug-in called GCodeWriter that outputs the g-code format normally. It is here: https://github.com/Ultimaker/Cura/tree/main/plugins/GCodeWriter

Which plug-in is used to write the file depends on the file extension the user saves the file with. If the user chooses ".x3g", the X3GWriter plug-in gets used. If the user chooses ".gcode", the GCodeWriter gets used.

`.pyc` files are Python code pre-compiled. They don't contain g-code. Some software can still analyse more or less what it's doing by the bytecodes. But it's a hassle, and open source software like Cura and this plug-in should always provide the .py files along.

a5421306, 2022-11-29:
尊敬的老师通过您的方法我成功的转换成gcode了！万分感谢您！但是gcode文件开始被故意添加了很多行没用的文件 ？
如下
M4010 X320 Y320

M4010 I0 T767 'ffff30088c713012ffdf8c713012ffff30a394b23022ce793003ffff3044ce59ffff300f8c713010ffff30038c71300fffff306f94b23021ce793003ffff3016ce59ffff3032ce59ffff30418c71300fffff30048c71300fffff3054'
M4010 I767 T855 '94b23021ce793003ffff3032ce59ffff30748c71300dffff30068c71300dffff303a94b23021ce793003ffff30c28c71300dffff30078c71300cffff301f94b23021ce793003ffff3035ce59ffff3066ce59ffff30418c71300b'

a5421306, 2022-11-29:
Dear teacher, I successfully converted to gcode through your method! Thank you very much! But the gcode file was deliberately added with many useless lines?

as follows

M4010 X320 Y320
M4010 I0 T767 'ffff30088c713012ffdf8c713012ffff30a394b23022ce793003ffff3044ce59ffff300f8c713010ffff30038c71300fffff306f94b23021ce793003ffff3016ce59ffff3032ce59ffff30418c71300fffff30048c71300fffff3054'
M4010 I767 T855 '94b23021ce793003ffff3032ce59ffff30748c71300dffff30068c71300dffff303a94b23021ce793003ffff30c28c71300dffff30078c71300cffff301f94b23021ce793003ffff3035ce59ffff3066ce59ffff30418c71300b'

Ghostkeeper, 2022-11-29:
You are not using this plug-in any more then, so this is getting off topic for the X3GWriter plug-in. Please take this up with the distributor of your version of Cura.